bchanudet/OctoPrint-Octorant #5: Not in Plugin Repository?

Patronics, 2018-03-10:
The readme.md says this plugin can be installed through the plugin repository, but I can't find any sign of it when searching relevant terms in the repository. Is it in the repository under some other name, or not actually in the repository? Is there any disadvantage to installing it directly from the link given (perhaps not autoupdating, etc?)

bchanudet, 2018-03-11:
Hello and thanks for your question!

You're right, the plugin is not in the plugin repository of OctoPrint. As I was not aware of anyone else interested of this plugin, I didn't do it at the time. But your feedback proved me wrong, so I took the time to check the plugin compability with the latest version of OctoPrint and finally create the pull request needed to integrate the plugin into the repository. I'm currently waiting for the merge of the PR : https://github.com/OctoPrint/plugins.octoprint.org/pull/172 Hopefully this will come soon and you will be able to download it from the plugin manager directly inside OctoPrint.

Installing the plugin from the link given does prevent OctoPrint to perform auto-updates, but that is the only disadvantage that I know of.

Don't hesitate me to ask me any other questions :)

foosel, 2018-03-12:
> Installing the plugin from the link given does prevent OctoPrint to perform auto-updates,

Actually, updates will also work if not installed from the repository, as long as your software update configuration is configured correctly (your's looks fine). Once you [actually publish a release here on Github](https://github.com/bchanudet/OctoPrint-Octorant/releases) it will be picked up.

The repository has nothing to do with software updates, that's all in your own hands ;)

bchanudet, 2018-03-12:
@foosel Thanks to you for stopping by and correcting my assumptions. Congratulations for OctoPrint by the way, that's a very fine piece of software (easy to extend and with good documentation) and this is a lifesaver for managing and using a 3D printer. :)

I wasn't sure how the plugin manager detected an update to a plugin, and I hadn't read anything about it in the documentation, but using releases makes a lot of sense. I'll just have to remember it when I fix a bug and want to deploy it for everyone!

closed 2018-03-12 by bchanudet

foosel, 2018-03-12:
Take a look at [the docs of the plugin here](http://docs.octoprint.org/en/master/bundledplugins/softwareupdate.html), that might help clarify things a bit :)