bchanudet/OctoPrint-Octorant #6: Suggestion for improvement

KukyMan, 2018-03-12:
Hey, it would be great if you add a command to the bot, which will cancel/pause the print. So if you will be outside printing some long print, you look on the snapshot and realize that it failed, you just type like !cancel or !pause and it will stop the print. But I love the idea of using discord as a "public server", so you do not have to create vpns etc.

blazze11, 2018-03-12:
It would also be nice to add something like "!snapshot" to create a snapshot and send it to the discord server.

bchanudet, 2018-03-12:
TL;DR : having commands is basically outside the scope of this plugin, but I'm open to some collaboration on a new plugin for handling a bidirectional system between Discord and OctoPrint. :)

--- 

Hello to you two and thanks for your suggestions.

Actually this plugin is not a real bot like Erisbot or Mee6. A real bot (as Discord consider them anyway) is a machine that receives all the messages from one's Discord server, and does an action based on keywords or commands. 

For now this plugin use a low-effort API endpoint made available by Discord just for sending messages, outward-only as I assume a lot of Octoprint's instances has access to the Internet but are not accessible from outside and are well hidden behind firewalls or proxies.

Making this plugin as a "full bot" able to receive commands would require some work as:
- A machine that I maintain myself, where I would receive all the messages from your servers, and would access to your own OctoPrint instances to issues the commands like !cancel or !status.
- Or ask the users to host the bot, by installing whatever tech stack I decide to use for this and make them create a developer account on Discord and go through the Developers documentation to create their own versions of the bot.

In both cases, the OctoPrint instance will need to be accessible from the Internet and the user will have to go through some firewall/proxy/VPN settings to make it so. So in a way, the user will already have access to OctoPrint from outside and will be completely able to cancel/pause the print from there, especially with the [OctoPrint-TouchUI](https://github.com/foosel/OctoPrint-TouchUI) plugin.

As I am quite concerned by my (and other's) privacy, and especially regarding the upcoming [GDPR](https://www.eugdpr.org/), I'm not comfortable having a server under my responsibility, where I would receive thousands of messages from unknown users and unknown servers, alongside storing access information to public OctoPrint instances, and even less issuing commands to them. Even with all the warnings and discharge declaration I could add, I could still receive an angry message from someone that had a fire accident in their garage because the OctoPrint instance received an unwanted command.

So only the second option is remaining. For now I don't see users ready to go through installing and configuring their own bot servers only to manage an OctoPrint instance, but if you can convince me otherwise, I could have a look at the full Discord API, and maybe create another plugin/repository to handle this.

Until then, I prefer keeping this plugin completely on the way "from OctoPrint to Discord", as this is primarily my use case. I use this plugin because I often design and print pieces for my friends and this way they can see and follow the progress of the print. I myself stay pretty close to the 3D printer to follow it in realtime and be ready to act if necessary.

KukyMan, 2018-03-13:
Oh, I am sorry to hear that it is so complicated.

closed 2018-04-04 by bchanudet