bchanudet/OctoPrint-Octorant #7: Webcam image is reversed

waumand, 2018-03-12:
Hi,
your plugin is really cool (maybe one of the best Octoprint plugin I used !), but sadly it didn't keep the reversal statut of your webcam.
My camera (official from Raspberry Pi) is verticaly reversed and it's fine with Octoprint but not right with your notification picture in discord as you can see with my picture.

[image]
Can you fix it ? Thank's a lot !

bchanudet, 2018-03-12:
I figured out this is not a real duplicate, as #8 is actually the newest of the two bugs, but to make it easier I'll update the progress there. Your comment is duly noted though, thanks for your feedback!

bchanudet, 2018-03-14:
As said in #8, I just release [version 1.1.0](https://github.com/bchanudet/OctoPrint-Octorant/releases/tag/1.1.0) that gives the plugin the access to these new parameters so it can send the snapshot in the expected orientation.

closed 2018-03-14 by bchanudet