bchanudet/OctoPrint-Octorant #8: RPi Cam upside down - not turned when sending to discord

Jan2220, 2018-03-12:
Hey,
The RPi Cam of my printer sits upside down and ive choosen in the octoprint settings to flip it correct. The OctoPrint Telegram Plugin seem to take these options too but i think you didnt added this feature?
Hope you can add this feature so i can view the picture without trying to flip my head instead 😆

waumand, 2018-03-12:
I got exactly the same trouble (see my issue). thank's for your great job btw !

bchanudet, 2018-03-12:
Hello and thanks to you two for your feedback,

Indeed I hadn't thought of these cases, as it is on the right side for me. I will work on it, in fact I had the time since the development of this plugin to have a look at the official image processing module for Python, so it should be (quite) straightforward...

Stay tuned!

nangad, 2018-03-13:
I have the same issue. FYI this plugin is AWESOME!!!!!!

cristianefe, 2018-03-13:
This works perfectly :)
> ssh into the pi and edit /boot/octopi.txt. it is commented in there so it should be straightforward. at the line that says: camera_raspi_options="-fps 10", you need to add -hf or -vf for horizontal or vertical flip.
> 
> if you are unsure how to do that, remotely poweroff the raspberry pi through the UI, pull the SD card and put it on your desktop computer and edit the text file that way.

From:
https://github.com/anoved/OctoPrint-EmailNotifier/issues/8#issuecomment-187017056

bchanudet, 2018-03-14:
This suggestion has been done and was delivered in the [1.1 release](https://github.com/bchanudet/OctoPrint-Octorant/releases/tag/1.1.0). Hopefully your OctoPrint instances will pickup the new release and display a popup to you for updating the plugin.

Don't hesitate if you encounter any trouble. :)

closed 2018-03-14 by bchanudet