bchanudet/OctoPrint-Octorant #10: Custom Snapshot URL?

TehEngineer, 2018-03-13:
I'm using an android phone streaming it's camera feed over the LAN Network for the camera feed, as opposed to the server's own camera server. Octoprint is configured to refer to it as the camera feed. Is there a way to use a/the custom image server to send images to discord?
Great plugin so far, by the way. I'm happy to see a discord version of an octoprint notifier.
[image]

bchanudet, 2018-03-14:
Hello and thanks for your feedback!

Actually, OctoRant doesn't use the Stream URL at all to get the snapshot, but rather the **Snapshot URL** field that is under the "Timelapse" subcategory. 

The main difference is that snapshots are grabbed by the computer where OctoPrint is installed, whereas the stream is displayed directly to the user and thus should be accessible from the user's computer. 

Actually, a bit like you, instead of using a camera server installed on my OctoPrint computer, I'm using a Wifi IP camera. The settings are looking like this:
[image]

The fact that you actually had a picture sent with your test message tells me you already have something setup in this field but for some reason it returns a black snapshot instead of the actual view. Incidentally that also means that if you have enabled timelapses, those are not working either. I suggest looking at this first. If you are not sure what to look for, please tell me and we'll figure it out. :)

bchanudet, 2018-05-13:
Hello,

As there was no answer to this issue, I'm closing it. Feel free to reopen it if necessary.

Cheers.

closed 2018-05-13 by bchanudet