bchanudet/OctoPrint-Octorant #11: Feature Suggestion: Add option to give time-based progress updates

Patronics, 2018-03-15:
It would be useful to have an option for time based progress updates, for example on very long prints an update every (for example) 10% would not come often, while small prints would have constant notifications. It would be great to be able to set it to (for example) every hour give a progress update if it hasn't given one during that periopd based on hitting a target percentage (or have it run entirely independently/instead of percentage-based progress updates). 
Based on my (limited) research of the OctoPrint plugin system, it seems you may need to get the time from another source than the payload data passed by the progress/event hooks.

bchanudet, 2018-03-17:
Hello and thanks for your feedback (and for your PR too!)

Time-based progress updates can indeed be a good thing to have regular update no matter how long the full printing process is gonna take.

I also experienced a bug in the last days where I relaunched a print of an updated file, but OctoPrint didn't reset the percentage according to the new file size, and I ended up getting a 125% notification...

There is two ways I can look into this:
- Really be time-based, so you can configure if you want X screenshots per hour. For long prints with constant average time between layers it could be a good solution. It may need for me to start a background thread and check to not interfere with the rest, but why not.
- Or implement a progress notification based on the Z-change (that is already provided by OctoPrint), so you could configure to receive a notification every X cm (for example every 2 cms). The advantage of this way is that you would get notifications proportionally to the "layer printing time". 

Maybe I could offer the three solutions, and the user would have to select one (or more) of them, depending on their notification preferences.

What do you think?

Patronics, 2018-03-17:
Thanks for replying, and I agree that either of those options would be great, and it’d probably be best to have both options 😀. That’s one thing I love about you plugin, how much configuration it allows for customization 😄.

cdub3d, 2020-02-19:
Hi, I just found out about this plugin today, and I had a similar question but for different reasons. I have a printer I do not control via Octoprint for various reasons, but I do have one running with the pi-cam. I use it to stream using youtube live plugin when I have to drive away for work when something is printed.   So I was wondering about the timed option. Not polling octoprint for any status of the printer/job at all... only sending snapshots based on time settings.

closed 2022-10-13 by bchanudet