bchanudet/OctoPrint-Octorant #14: Feature request: script support

Divinii, 2018-03-17:
Hey,
I have a relays installed on my printer to control several things like a led strip. Would it be possible to add functionality to support Terminal commands like controlling GPIO pins?
What I had in mind was that i´d be able to switch on the led strip before a snapshot is made and switch it off after the snapshot has been taken. Maybe with a delay option.

bchanudet, 2018-03-17:
Hello and thanks for your feedback!

This is actually an interesting request that could be "quickly" done. 

One thing though : would you consider it "good enough" if the terminal commands can only be added in the configuration file `config.yaml` and not through the configuration page in OctoPrint ? 

I'm thinking about this from a security view: the configuration page in OctoPrint is available to every administrator logged in, whereas the file `config.yaml` is editable only by the people that have a physical access to the machine hosting OctoPrint. So forcing the setup of such commands only from the `config.yaml` should be safer and would prevent any not-so-well-intended user to change the "before" script to a `rm -rf /`

What do you think ?

Divinii, 2018-03-17:
I´d think only making this option accessible through the config is a great idea that i haven't thought about. Since running terminal commands from any admin could actually be quite the security breach. Although i am not entirely sure if the webui has sudo access // or needs it for gpio actions (kinda a raspi noob here).

Also, a maybe more sleek option could be for the config file to only enable this functionality, so it is defaulted to disabled and only if you want this feature you could enable it. For example have the settings still in the octoprint ui with the hint to jump to said file and enable it if you´d want to use it.

bchanudet, 2018-04-04:
This suggestion was delivered in the latest version [1.2.0](https://github.com/bchanudet/OctoPrint-Octorant/releases/tag/1.2). Feel free to test it and tell me if you encounter any trouble. :)

closed 2018-04-04 by bchanudet

bchanudet, 2018-05-02:
Hello!

I willingly admit the small message is not clear enough. I was meaning to add a Wiki entry on this project about this but I haven't found the time for this yet.

The `allow_scripts` flag is to be put in the OctoPrint configuration file, as you guessed it correctly, `octoprint.yaml`. The "trap" here is that the flag **doesn't exist if it is not manually written**, which is why you didn't find it.

It's actually from (I suppose, I haven't checked thoroughly) OctoPrint's behaviour around handling configuration file: OctoPrint doesn't store values that are not different from the default values, surely to keep the configuration file small. 

So, to set the flag correctly, you have to put the `allow_scripts` value at the correct place. If you've already touched the configuration for Octorant, you should find an Octorant section in the `octoprint.yaml` file. Keep in mind that .yaml files, much like Python itself, uses indentation for defining sections.

An example:
```yaml
plugins:
  octorant:
    events:
      startup: 
        message: Yay, I'm happy!
        with_snapshot: true
```

The easiest way is to add `allow_scripts` right under the `octorant:` line: 
```yaml
plugins:
  octorant:
    allow_scripts: true
    events:
      startup: 
        message: Yay, I'm happy!
        with_snapshot: true
```

Then, once the file was saved, you can restart OctoPrint, and the configuration for the scripts should be usable. If not, it can be a bug and I will gladly try to help you solve it. :)

Cheers!