bchanudet/OctoPrint-Octorant #15: Not sending message

RedWireAP, 2018-03-21:
I've noticed that without a webcam setup, and if the include snapshot boxes are checked, messages to discord will not be sent.

bchanudet, 2018-03-21:
Hello, 

I'm very sorry hearing the plugin doesn't seem to work on your OctoPrint instance. 

However, the plugin already checks if the webcam is correctly setup (that the field **Snapshot URL** under **Webcam & Timelapse** is field with a value starting with `http`), and if the screenshot couldn't be grabbed nonetheless, the message is sent to Discord anyway. 

Of course there is always the possibility of me missing something, but after trying on my instance to reproduce the bug, I didn't find any way to do it : no matter what, I always receive atleast the text message in Discord.

So I guess I'll need more information from you if possible, so I can know where to investigate : 

- if you uncheck the snapshot checkboxes, do you receive the messages? 
- if not, try changing like the name of the Bot (second configuration field for OctoRant) and click `Save`. Did you receive the test message? 
- did you double-check the Webhook URL you got from Discord?                
- do you see any errors on the console running OctoPrint? If you can't see the console (e.g. you're running OctoPi or OctoPrint is running as a service), can you provide me the file `octoprint.log` that you can get from the **Logs** page in the configuration. If you worry about posting your own logs publicly on this issue feel free to send them to my e-mail hello@benjaminchanudet.com
- what configuration do you have for **Webcam & Timelapse** in OctoPrint? Does the button `Test` on the **Snapshot URL** field works?

Thanks in advance for your help, hopefully we'll find what's troubling your OctoRant :)

RedWireAP, 2018-03-21:
1) Unchecking the snapshot box does then allow messages to be sent
2) changing the name of the bot (or any field) no test message is sent
3) Webhook URL is valid
4) email with log forthcoming
5) stream URL field is blank (and Test button is greyed out). I have yet to setup a webcam, i was thinking of doing so this evening or tomorrow so the fields are currently the Octoprint defaults.

Side note, i think this plugin is a great resource and i applaud you for creating it :) I'm happy to be of assistance in any way.

bchanudet, 2018-03-21:
Thanks for your quick and detailed response. I wish I could receive the same bug reports at work ;)

The logs helped me reproduce the bug and find the reason for it. I will have a fix coming soon. 

Until then, there is a workaround that should work on your side: 
1. Go to **Webcam & Timelapse** configuration
2. Set *anything* in the **Snapshot URL** field. Anything really, something like `test` or `fix a bug from a bad developer` if you want. :)
3. Save the settings.
4. Reopen the settings, change the name of the bot to trigger a test message to Discord.
5. If you receive it, you should be able to clear the field **Snapshot URL** to its default value.

Tell me if this solves the problem for now! And hopefully with the next version to come we won't have to fight this again :)

RedWireAP, 2018-03-21:
The workaround worked exactly as you described. I am now receiving the test messages. Cleared the Snapshot URL field and still receiving test messages.

Sorry to find a bug, but happy i was able to assist! And thank you again for your work :)

commit 5d55b040 (2018-03-21): Fix #15 : global_get() returns None when path is not set yet.
closed 2018-03-21 by bchanudet