bchanudet/OctoPrint-Octorant #16: Settings Panel

OctopusProteins, 2018-03-26:
When OctoRant is installed and enabled, all the settings panels for plugins are combined. After posting an issue in foosel/OctoPrint, we determined that this plugin was causing the issue.
https://github.com/foosel/OctoPrint/issues/2519

https://i.imgur.com/X3C7gw1.png
https://i.imgur.com/wkTIEMb.png
https://drive.google.com/open?id=1GvaEmDDZnLOWGUmIjNz04YlsggvIwcJ4

BillyBlaze, 2018-03-26:
After a quick look, seems like @bchanudet  made a typo `</string>` instead of `</strong>` on https://github.com/bchanudet/OctoPrint-Octorant/blob/master/octoprint_octorant/templates/octorant_settings.jinja2#L71

OctopusProteins, 2018-03-26:
@BillyBlaze Thanks. I saw this, but I'm not sure if it will fix the issue. Only one way to find out.

bchanudet, 2018-03-26:
Hello and thanks for your feedback!

Indeed I was able to reproduce the bug and to fix it thanks to your PR. So I merged it right away. If you uninstall and re-install the plugin, you should not encounter the bug anymore.

What I need to fix is also the fact that the plugin manager is downloading from the master branch instead of downloading the latest release currently available. After releasing 1.2.0 I will switch to a dev branch instead at least.

Thanks again for you PR :)

OctopusProteins, 2018-03-26:
Fixed by #17.

closed 2018-03-26 by OctopusProteins