bchanudet/OctoPrint-Octorant #18: Feature Request: Terminal Chat Channel

ear0wax, 2018-03-27:
Title says it all. Make a text chat act as the serial terminal.

bchanudet, 2018-03-27:
Good evening,

I'm refraining myself to close the issue based solely on the complete lack of politeness. 

Even if the title should *say it all*, I can't even figure out what your request is and how different it is from #6 where I already said I won't try to make this plugin able to receive input from a Discord Channel. 

If I failed to grasp your idea from the merely 18 words you gave me, feel free to tell me more about it.

Best regards.

ear0wax, 2018-03-28:
I'm glad that you feel that the content of your comments doesn't determine it's politeness; but just the word count. Wishing your project the best. (25)

closed 2018-03-28 by ear0wax