bchanudet/OctoPrint-Octorant #19: If the camera stream is not setup right, it wont send a message at all.

cameroncros, 2018-04-07:
I have got octorant setup without a webcam setup properly.
When it octorant tries to send a message (with "Include Snapshot" on), the URL for the snapshot times out, and the message is not sent. It ideally should still send the message, perhaps with a warning that the camera is not setup right?

bchanudet, 2018-04-08:
Hello and thanks for your feedback.

This looks somewhat close to what an other user found out in #15 and that should have been fixed with the commit 5d55b040e2b7187d57d8360c206def3235a28b1a. 

But as you indicate that the URL times out, should I understand that you actually got a URL in the webcam setup, but this URL is not working properly? Or that the webcam URL were not setup at all? 

In any case, would you mind sending me a log of your OctoPrint instance? The easiest way to get it is grab the file `octoprint.log` in the **Logs** section of the configuration page of OctoPrint. You can either attach it to this issue, or if you prefer send it to my e-mail address hello@benjaminchanudet.com.

Hopefully this will help me finding out the cause. In the meantime I will have a look at how a faulty URL is handled in Octoprint's timelapse feature.

cameroncros, 2018-04-09:
My camera was not configured properly, so the URL was a dead URL (mjpg_streamer was not running on the port). Octorant then tried to get a snapshot, which timed out, and that caused the entire message to be aborted. I will get you a stack trace shortly.

cameroncros, 2018-05-27:
See: https://github.com/cameroncros/OctoPrint-DiscordRemote/commit/00464942fc16e32a1f2e3bbe99ae3caca0e8796a
Cherry pick may not work, but something like that might work for you?

commit 8c5da296 (2018-06-09): Fix #19 (finally!)
closed 2018-06-09 by bchanudet