bchanudet/OctoPrint-Octorant #20: Feature Suggestion

cameroncros, 2018-04-09:
It would be cool if the messages could contain variables. 

For example:
* Startup Message: "⏰ I just woke up! What are we gonna print today? Im at %IP_ADDR%" 
  * %IP_ADDR% gets substituted with the systems IP address, so the resultant message is:
"⏰ I just woke up! What are we gonna print today? Im at 192.168.1.123"

bchanudet, 2018-04-18:
Hello and thanks for your feedback. 

I will look if I can quickly grab the IP of the Octoprint Instance and offer it as a new variable that can be used in the messages.

Stay tuned :)

cameroncros, 2018-05-27:
Have a look at: https://github.com/cameroncros/OctoPrint-DiscordRemote/commit/383b56786dcc6a460bd469fcef128ae7ea705312
Maybe you can cherry-pick this change? Its not perfect, as it doesnt deal with port forwarding, but might be useful to start with.

noxin1, 2019-08-17:
+1 for this feature.     I have a few octopis across the network and I keep loosing them in the network.

closed 2023-05-31 by bchanudet