bchanudet/OctoPrint-Octorant #21: Feature Suggestion: Interact with printer from discord

cameroncros, 2018-04-09:
Allow users to control the printer via chat messages.

Not sure if its possible, but allow the users to send messages to the discord channel and have the plugin respond to them.

For Example:

"Cancel Print" -> cancels the print
Attached file to chat, -> uploads file to octoprint
"List files" -> lists all files available for print
"Print [filename]" -> starts the print of a file
"Power on" -> turns on PSU (Might require intergration with https://plugins.octoprint.org/plugins/psucontrol/)
"Power off" -> turns off the PSU

This is something I will likely implement myself at some point.

bchanudet, 2018-04-18:
Hello again!

This feature had already been asked in the past (see #6) and to be honest, I don't see me having enough time in the next months to handle such a feature.

Also I think this feature could be a whole plugin by itself that could be independent from mine :)

If you start working on this, feel free to ask any question you may have, though I'm really not an expert in Python programming.

closed 2018-04-18 by bchanudet

cameroncros, 2018-04-19:
I have written most of the discord bot api necessary to complete this, but still needs some work. Ideally, if we had python3, we could just directly use discord.py, but my code will do til then. Once I finish, I will create a merge request back and you can decide if you want to keep it. Might be done in the next few weeks.