bchanudet/OctoPrint-Octorant #24: URL Snapshot in Motion?

carlymx, 2018-04-27:
**ENG(GTraslator)**

Hi.

First of all, Magnifico addon, keep it up.

I use [Motion](https://motion-project.github.io/index.html) to manage my USB camera for Octoprint without problems.
Now I want to add the snapshot that is saved by default in `/var/lib/motion/lastsnap.jpg` and I do not know how to get it because I can not find (if any) a route to that snapshot.

`192.168.0.100: 8081/0` shows the image in Streaming and is not valid.
`192.168.0.100: 8080/0` you access the WebControl and it is not valid either.

One thing I tried unsuccessfully (I do not know why) is to send the snapshot to a directory where Octoprint takes the images for example `./Static/img/` if I visualize the full URL in the browser for example `http://192.168.0.100:5000/static/img/watermark.png` we visualize the image perfectly but when adding it to the `Snapshots` section of the Octoprint it gives an error, but in the Stream URL section it works.

You can help me, any solution would be worth it.

**ESP.(Original)**
Hola.

Antes de nada, Magnifico addon, sigan así.

Uso [Motion](https://motion-project.github.io/index.html) para gestionar mi cámara USB para Octoprint sin problemas.
Ahora quiero agregar la instantánea que se guarda por defecto en `/var/lib/motion/lastsnap.jpg` y no se como conseguirlo por que no encuentro (en caso de existir) una ruta a esa snapshot.

`192.168.0.100:8081/0` muestra la imagen en Streaming y no vale.
`192.168.0.100:8080/0` accedes a la WebControl y tampoco vale.

Una cosa que he intentado sin éxito (no se por que) es enviar la snapshot a un directorio donde Octoprint coja las imágenes por ejemplo `./static/img/` si visualizo la URL completa en en navegador por ejemplo `http://192.168.0.100:5000/static/img/watermark.png` visualizamos perfectamente la imagen pero al añadirla al apartado `Snapshots` del Octoprint da error, pero en el apartado Stream URL funciona.

Pueden ayudarme, cualquier solución me valdría.

[image]

cameroncros, 2018-05-28:
This thread has some information on how to get a single image served up by motion:

https://groups.google.com/forum/#!topic/octoprint/BIzJdZmqvmk

Might be the easiest option? Alternatively, OctoRant could be changed to read the file from the disk, but I dunno how useful that would be though.

OctopusProteins, 2018-08-12:
Sorry, ID number referencing a different commit.

megasaturnv, 2019-03-14:
Saw this issue and if it helps anyone, here's a gist I made a while ago in PHP to output a snapshot from a motion stream (Or any mjpeg stream if setup correctly). Quite a nice solution as it doesn't require any temporary files or proxying.
https://gist.github.com/megasaturnv/81279fca49f2f34b42e77815c9bb1eb8

There's also a slightly different version which puts the snapshot in an \<img\> tag.
https://gist.github.com/megasaturnv/35578337662acd28e0bcd2946a4b069e

donbarthel, 2020-06-21:
I have the same issue, streaming from Motion works but snapshots do not. It would be good if the snapshot URL could be a path instead since Motion and OctoPrint are running on the same Pi.

meteozond, 2020-09-05:
Another one approach with nginx content_by_lua https://gist.github.com/meteozond/b2c07c850fdcc7f41ec065756ccfb2e0

van-ant, 2022-04-04:
here is a quick and dirty solution that works.(well not so quick, havent done any programming for a while)
copy the script below to a file "octosnapmotion.py" for example
run it with python3 octosnapmotion.py

set the timelapse url in octoprint to http://127.0.0.1:8082

bingo!

zackfuchtel, 2023-01-06:
> here is a quick and dirty solution that works.(well not so quick, havent done any programming for a while) copy the attached file run it with python3 socket_server.py
> 
> set the timelapse url in octoprint to http://127.0.0.1:8082
> 
> bingo! [socket_server.zip](https://github.com/bchanudet/OctoPrint-Octorant/files/8417868/socket_server.zip)

Thank you.

Enoril, 2023-05-20:
You can use {IP}:{port0}/{camid}/**current** and obtain the image needed for the time-lapse
Or /**source** if you don't want the bottom left & right texts

cf the [documentation ](https://motion-project.github.io/motion_config.html#OptDetail_Stream)

```
[...]
    {IP}:{port0}/{camid}/stream Primary stream for the camera
[...]
    {IP}:{port0}/{camid}/source Source image from the camera
    {IP}:{port0}/{camid}/current Static JPG for the camera
```