bchanudet/OctoPrint-Octorant #25: OctoRant stalls printer leaving hotend on.

TheBuzzAldrin, 2018-05-02:
Using the latest nightly build of Octopi and latest OctoRant release. 
After enabling plugin and adding the webhook I started a print and everything continued normally for about 20% of the print until Octoprint gave me a serial error and the printer stalled mid print. At the same time a user in my Discord server had sent a message in the same channel OctoRant sends it's status updates. This leads me to believe that OctoRant is not handling other messages in the same channel correctly and could lead to a dangerous situation should the hotend remain on and unchecked.
I reran the same .gcode through Octoprint with OctoRant removed and it finished successfully. Any help to resolve this would be amazing.

TheBuzzAldrin, 2018-05-02:
Should also note I'm using the Original Prusa i3 MK3.

bchanudet, 2018-05-02:
Hello and thanks for your feedback,

At first, I will be clear on one very important point: this plug-in is very limited on what it does and I am doing everything possible to keep its scope as minimal as possible.

That means two things:
- Octorant **does not send** any commands to the 3D printer at all. On the OctoPrint side, the plug-in is only fetching regularly the state of the printing, and that's it.
- Octorant *can not read* messages from a Discord channel. On the Discord side, I am using a hook provided by their Sdk that **only allows for sending messages**. For reading messages the setup instructions would have been much longer, as you would need to invite a bot and so on (and I would force you to read a loooong privacy terms document to cover my ass for getting all your messages ;))

Starting from this, as I can't act on the printer, I don't see any real reason for this plug-in to be the reason of your faulty print, even less coming from a Discord message.

Nevertheless I would be interested on seeing your logs from OctoPrint, if there is an other error on which I can help you, even if unrelated to my plug-in. For now, I see it as OctoPrint saw it: a serial port error.

Feel free to give me any information should you have more. :)

Cheers

TheBuzzAldrin, 2018-05-02:
I will update this afternoon with the log, I am unable to retrieve it while I am at work. Thank you for your patience. 

Thank you for your response. I was also confused as to how it caused it but repeat testing leads me to believe it is. Files ran on the same Octoprint installation without Octorant (only 3rd party plugin installed at the time) finish successful with no issue. As soon as OctoRant was added back and a message sent mid print to the discord channel the printer would stop in it's place leaving the hotend on and Octoprint displaying an error until the MK3 was reset and a connection re-established. 

I do appreciate the response and am anxious to hear back once I post my logs.

bchanudet, 2018-05-02:
No problem, I can wait for the log. If you worry about leaving it in public in this issue, you can also send it to my e-mail address hello@benjaminchanudet.com

By the way, I have a some other questions for you, so I can investigate further:
- On which platform is OctoPrint installed? 
- Is Discord installed (and running) on the same computer that is hosting OctoPrint?
- Do you have other plugins installed in OctoPrint?
- What is your general configuration of the plugin? (e.g. which messages are sent, with snapshots, do you use the script feature, etc)

Hopefully we'll find the reason for this problem. :)

TheBuzzAldrin, 2018-05-02:
1. I have my Octoprint server running on a dedicated RPi 3. No other software except Octoprint. 
It is a fresh install as of yesterday done using the latest nightly build.
2. Discord is not on the RPi as mentioned above, the messages were being viewed on the Discord client from a separate computer running W10. 
3. No plugins except the default ones included with Octoprint.
4. Only configuration done was to change the fields regarding webhook URL and bot name. 

I will be uploading the logs here shortly, I will be home in about an hour. 
Thank you for taking the time to help me diagnose my issue.

TheBuzzAldrin, 2018-05-03:
Gonna go ahead and close this issue. Attempted to replicate the same issue with the same test I used last night, but no issues presented themselves. Must have been a couple one off issues, I apologize for any worry. Thank you for your time with helping me, but I am unable to replicate the issue anymore. 
Will reopen if there are any further developments.

closed 2018-05-03 by TheBuzzAldrin

bchanudet, 2018-05-03:
Purely on the developer side of this plug-in, I'm glad to hear it's not my code that caused the issue. 

However as a 3D printing fellow, I know that these problems are stressful, especially when it's on a big print running for hours. I hope all is fixed and you will enjoy your printer :) 

If you have any other issue, feel free to reopen the issue and we'll look together into fixing it ;)