bchanudet/OctoPrint-Octorant #26: Codestyle not conformant with PEP8

cameroncros, 2018-05-06:
The indent style is set as tabs in .editorconfig for python files. PEP8 suggests using 4 spaces instead of tabs. Is the usage of tabs intentional, and if not, can it be migrated back to spaces?

bchanudet, 2018-05-09:
Hello and thanks for your feedback!

The `.editorconfig` file is actually coming from the [cookiecutter template](https://github.com/OctoPrint/cookiecutter-octoprint-plugin) for Octoprint's plugins. I didn't modify it at all.

However, PEP8 also suggests that indentation should be consistent with existing code, and it seems that OctoPrint also [uses tabs for indenting](https://github.com/foosel/OctoPrint/blob/master/.editorconfig) so to be honest with you I don't think I will change it, in the case that OctoPrint will get someday migrated to Python 3.

I hope you'll understand my position on this :) 

Cheers,

cameroncros, 2018-05-09:
Yeah, i have to admit i read the octoprint .editorconfig as well, and saw that. I am irritated, but I guess its not fixable.

closed 2018-05-09 by cameroncros