bchanudet/OctoPrint-Octorant #31: Not able to set 99% as notification threshhold

banditb17, 2018-06-20:
when my print is done, it moves out of the frame of the camera so you only get a partial shot. I tried to set it "notify every 99%" hoping it would take the picture just before the print is done, but it just failed to notify at all. What are my options?

Leapo, 2019-03-21:
I'm having a similar issue. My end-gcode includes a command to shut off the printer's lighting, which means my last photo is very dark. I need a way to trigger the "complete" photo before M81 is sent to the printer.

bchanudet, 2019-03-24:
I just figured that I never answered to the original issue from @banditb17, sorry for this!

I must admit that as I'm not in this case, I don't really know what I could do to suit your needs. But maybe you have some suggestions about what you'd like to have in the plugin to manage this case?

Leapo, 2019-04-11:
How about having the option to take a snapshot when a specific M117 command is sent to the printer?

For example, trigger a screenshot on "M117 Snapshot"

Then we can have a photo taken as part of our end gcode.