bchanudet/OctoPrint-Octorant #32: Estimated time to finish

ProfEngr, 2019-03-11:
Can an estimated time remaining be added to the percent progress messages? Or did I just miss it going through the docs? The pushbullet plug-in has the option, but I've been using yours more lately.

bchanudet, 2019-03-11:
Hello,

You didn't miss anything in the docs, the percent progress messages don't include the remaining time currently. But I had a go this evening about it and was able to get something usable.

Like we already have `{time}` and `{time_formatted}` available for the time spent, would `{remaining}` (for the raw value in seconds) and `{remaining_formatted}` (value formatted as HH:MM:SS) suit your needs?

ProfEngr, 2019-03-11:
That would be great. Thank you.

bchanudet, 2019-03-11:
I just published a new release, **1.2.2**, that should hopefully meet your expectations. :)

The update should appear in the plugin manager of your OctoPrint instance.

Feel free to report any issue if you shall encounter any :)

closed 2019-03-11 by bchanudet