bchanudet/OctoPrint-Octorant #34: Mistyping a key causes error

StefanJanssen95, 2019-03-14:
I accidentally mistyped a key in the format string and did not get any progress in Discord during my print.

**Expected Result:**
Just log the output string with the mistyped key. This makes debugging the cause easier and will still give you the correct feedback.

**Actual Result:**
When using an incorrect key python throws an error

**Info:**
```2019-03-14 18:26:17,179 - octoprint.printer.standard - ERROR - Exception while sending print progress to plugin octorant
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/printer/standard.py", line 223, in call_plugins
    plugin.on_print_progress(storage, filename, progress)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octorant/__init__.py", line 204, in on_print_progress
    self.notify_event("printing_progress",{"progress": progress})
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octorant/__init__.py", line 252, in notify_event
    return self.send_message(eventID, tmpConfig["message"].format(**data), tmpConfig["with_snapshot"])
KeyError: u'remaining-formatted'```

bchanudet, 2019-03-17:
Hello and thanks for your feedback! 

I will add an exception handling code to detect the mistyping of message variables and send the string as-is if the exception is triggered.

Hopefully I'll have time to do this quite soon.

bchanudet, 2020-03-18:
As it seems "quite soon" is literally one year and 2 days... Sorry for taking so long!

The bug is fixed, and the resolution described in #38. It will be available in the next version. :)

closed 2020-03-18 by bchanudet