bchanudet/OctoPrint-Octorant #43: @mentions not working?

PitbullRaven, 2020-05-06:
Hi. I have tried getting Octorant to use @mentions, without success. 

For instance, for the printer error state, in the Octorant settings, I have:
⚠️ @MyHandle  Your printer is in an erroneous state.

This results in that text showing in the stream, but the handle is not highlighted and does not trigger the ping.

I have also tried @MyHandle#6809 which would be my user according to Discord, yet still nothing.

Any insight?

ragnork2, 2020-05-16:
There's something about discord that doesn't let mentions work without specific functions in the code, this makes things like sudo commands not able to work with mentions.

KipLandwehr, 2020-05-26:
I have tags working with my Octorant setup. The trick is that you have to use your unique ID. The format is
`<@!#################>`. You can get yours by typing `\@yourhandle#6809` into a discord channel.

PitbullRaven, 2020-05-28:
> I have tags working with my Octorant setup. The trick is that you have to use your unique ID. The format is `<@!#################>`. You can get yours by typing `\@yourhandle#6809` into a discord channel.

Thanks for the tip. I'll give it a go.

closed 2020-05-28 by PitbullRaven