bchanudet/OctoPrint-Octorant #46: No Progress Messages

recrudesce, 2020-06-10:
I have OctoRant installed, and it will happily message my Discord channel when the printer is online/offline etc.  But it will never send a progress message even when set to notify at 1%.

I suspect this is due to me having PrimeTimeGenius or the DisplayLayerProgress plugin installed.

Any thoughts as to what I can check ?

recrudesce, 2020-06-10:
OK, they are coming through, they're just not at the % that's shown in Octoprint.
Like I'm getting a 30% complete message when the print is about 50% complete.

I guess it's pulling the progress from somewhere else than the data being displayed ?

Fank, 2021-02-22:
I noticed the same issue, that the percent which is posted is not the same percent i see from the bar in octroprint.
This is my best example, check the percent vs time vs timestamp:
Format: `📢 Printing is at {progress}% running for {spent_formatted}, remaining {remaining_formatted}`
[image]

I was using the https://plugins.octoprint.org/plugins/DisplayLayerProgress/ before, because i thought it may collide with the percent progress and removed it, but still same behaviour.

bchanudet, 2021-02-22:
Hi @Fank and thank you for your report :) 

I'll have a closer look at the code handling the progress of the print and do some tests on my side. I'm also thinking about implementing other ways to track the progress of a print job. It should all land in a 1.3.0 version, though I can't really give a release date for now.

Fank, 2021-03-04:
Not sure if helpfull but the worst example i had:
[image]

[CCR6SE_Plate_v9.gcode](https://drive.google.com/file/d/1TJgiDNUU1DD1gDKY4P1gY7dAAZSPzZ6y/view?usp=sharing)

Maybe helps debugging

closed 2022-10-13 by bchanudet