bchanudet/OctoPrint-Octorant #49: Stopped sending messages and not sure how to diagnose

Cylindric, 2020-10-28:
I've been using Octorant for a while now, and it's been sending completion messages to Discord just fine. If I try sending test messages now, those work fine too.
Recently however, it has stopped sending messages on completion. I'm not sure where to look for any logs or other information about what the plugin is doing, so can't really investigate.
Can I make Octorant output some log data somewhere? I suspect another plugin is interferring, but without any logging, it's hard to tell.

Cylindric, 2020-10-28:
I found how to enable debug logging. I added some more logging to the start of `on_event` and it seems that Octorant isn't even getting the events. 

It looks like a problem with a different plugin. I disabled the Tasmota plugin, and the `PrintDone` events come in again.
I'll raise this with them.

bchanudet, 2021-02-20:
Hello and so sorry for not answering before. 

I appreciate you took time to investigate and find the responsible code for this to happen. I've looked at the Tasmota plugin and was able to find [your issue on their repository](https://github.com/jneilliii/OctoPrint-Tasmota/issues/98). It seems that the issue was gone once the plugin was reinstalled.

I'll therefore close this comment, but feel free to add any additional note  :)

closed 2021-02-20 by bchanudet