bchanudet/OctoPrint-Octorant #50: Feature Request: Post Timelapse When done rendering

RainyKitsune, 2020-12-02:
Recently, Octoprint's Default Timelapse was changed to render into .MP4 instead of .MPG. As a result, Discord can now play Octoprint timelapses directly in the channels with no additional post processing to the timelapse file necessary. Would it be possible to allow Octorant to detect when the render of the timelapse is finished, and upload it as it's own command?

mavermc, 2021-02-01:
I 2nd this!

bchanudet, 2021-02-19:
Hello to you both!

I've been toying with the plugin this evening and I tried something around your idea. I have a Proof of Concept that uploads the timelapse created by OctoPrint, and displays it directly in Discord without the need to download it first. 

Though it seems that Discord has a hard limit of 8MB per file when using the WebHook method, no matter if the server has Nitro or anything like that. The default bitrate of 10000kbits produces insanely big mp4 (my test timelapse is 5.4MB for 6 seconds...) ; by tuning it down to 1000kbits it's quite better (227KB for the same duration)

I haven't done a lot of timelapses yet (though that may change, I ordered an endoscopic webcam to copy CNC Kitchen's closeup setup). So if you're used to do some, I would greatly appreciate your feedback on whether the 8MB limit is too harsh a constraint to make this feature really useful for you. In which case I may stop my investigation and look at other features to develop instead :)

RainyKitsune, 2021-02-20:
File size likely depends on timelapse settings and a few factors about the print itself. I personally use the Z Height timelapse setting, and prints end up being over or under usually depending on how I have the print laid out. I'd say about a little under half of my prints end up under 8MB on my setup, though it's usually well lit and on a 'stationary' build platform. To my knowledge, h264 video encoding, which is what .MP4 uses, is more efficient when multiple parts of the video don't change from frame to frame. My setup probably is more advantageous to compression than most others, since my camera is attached to the printer bed, and thus doesn't have the background changing consistently which saves on compression, but since not everybody has a setup that stabilizes the build plate like this, it's likely the majority of users would have higher file sizes for equivalent prints.

I'd still personally like the feature, but hearing the reason, I also wouldn't be upset if it was decided that development time was better spent elsewhere due to how niche the use case actually is and how few 3d prints would practically create a valid timelapse for the webhook limit.

bchanudet, 2021-02-22:
Many thanks for your feedback @RainyKitsune. 

Honestly I was expecting a much lower number than 50% of your timelapses under 8MB. That's a good enough number for me and I think I'll pursue the development of the feature.

As a fallback, if the file is over 8MB, I was thinking about offering the URL of the timelapse (provided I can retrieve it from OctoPrint's API) as a new variable `{timelapse_url}` to include in the message. The caveat is that it would most likely be a link limited to the printer's network, so it would be useless if you're out and checking the status on your phone, but it could work if your printer is in the garage and you're upstairs on your computer. Do you think it could be useful for you or not?

RainyKitsune, 2021-02-23:
To me personally, I'd say no. Anywhere I can access a computer on the local
network is anywhere I can type in Octopi.local and download from there,
regardless of the printer's location in the house. The closest use I can
think of is maybe downloading timelapses to my phone, but I already have
other means to do that anyway. Furthermore, my bot is on a Discord server
with other people who are not on my network that sometimes check in, so I'd
have to tell them that the link won't actually do anything for them, so I'd
likely not end up using the timelapse_url for those reasons.

closed 2022-10-13 by bchanudet