bchanudet/OctoPrint-Octorant #51: Feature Request - Allow to message user directly with '@user####' in the message

rileyschuit, 2021-02-19:
There may already be a solution out there but I haven't found it.  I attempted to do '@discordusername####' on the beginning of my message and it makes it literal in the message in my group.

Thanks in advance.  I absolutely love this plugin, keep it up.

bchanudet, 2021-02-19:
Hello and thanks for using OctoRant :)

You're not the first to report this kind of problem, and it seems that Discord did some work to harden the way mentions work within Webhooks. I suppose to avoid users being spammed by a unfriendly bot?

After some investigation, I found a way to push notifications to a specific user (`@here`  and `@everyone`  are still working).

First, instead of using the user handle, you'll need to get the ID of the User. A support page from Discord is available for this : https://support.discord.com/hc/en-us/articles/206346498-Where-can-I-find-my-User-Server-Message-ID-

Then in OctoRant settings, you'll have to use the following format: `<@user_id>`, where `user_id` will be the long number you'll get from Discord.

Here is a screenshot of what it looks like on my virtual setup:
[image]

You'll need the same kind of things to notify roles, using the format: `<@&role_id>` (note the `&` character used to distinguish roles from users).

The format is well hidden in the developer documention of Discord, I may write something in the wiki to help others in the future.

Hope this helps!

rileyschuit, 2021-02-19:
AWESOME!  Consider this closed.  Thank you so much.  Keep up the good work!

closed 2021-02-19 by rileyschuit