bchanudet/OctoPrint-Octorant #58: Upload to SD Printing Process Bug

SalmonSays, 2022-08-05:
I have just recently started uploading files to my SD card through Octoprint after noticing much better print quality using this method, and a quirk appeared in OctoRant on the latest version of OctoPrint as of time of writing- the Upload to SD prompt uses the same percentage progress bar as a normal print process, and it sends Printing Progress messages out with screenshots even though no print is in progress. Instead screenshots/messages are posted at SD card file upload intervals matching the "notify every xx%" print progress interval.

omnislash772, 2022-09-18:
Confirming that I am also getting this issue
[image]

omnislash772, 2022-09-19:
Should be as easy as suppressing any updates if octoprint's state is "Sending file to SD"

commit 622549e4 (2022-09-20): Fix #58 by creating new events dedicated for SD cards transfers
closed 2022-10-13 by bchanudet