bchanudet/OctoPrint-Octorant #62: Prevents OctoPrint from restarting in a timely manner

foosel, 2022-10-18:
I recently updated to 1.3.0 (which btw is in an update loop since setup.py says 1.3.0 but the tag says 1.3.0.2) and since then noticed extremely long restart times after updates or plugin changes. Took me a while to narrow it down to OctoRant, but disabling just that plugin made the issue go away. Usually, a problem like that is caused by failing to set `daemon=True` on a spawned Thread, since that will then keep the whole process running even if the main thread has exited. A quick look at the source indicates that `DiscordMessage` might be the culprit here since it is not being initialized as a daemon thread.

You can learn more about daemon threads here: https://docs.python.org/3/library/threading.html#thread-objects

bchanudet, 2022-10-18:
Hi @foosel,

Many many thanks for the time you took to investigate the issue. I admit not being a specialist regarding Python's threads, and I thought that setting `daemon=True` on the thread would actually prevent OctoPrint to shutdown correctly. 

I'll have a look on it this very evening and release a new version. That will also fix #63

foosel, 2022-10-18:
Glad to hear I'm not the only one getting confused by this 😊 When I started out with OctoPrint, I also misunderstood what "daemon" meant at first. Took me some head scratching to get this right, since then it's one of these issues that when you see it happening you immediately recognize it after making the mistake yourself 😅

commit b0fdb986 (2022-10-18): fix #62 : declare DiscordMessage thread as daemonic to avoid trouble with OctoPrint
commit c2b2d493 (2022-10-18): fix #62 : declare DiscordMessage thread as daemonic to avoid trouble with OctoPrint
commit 07888d87 (2022-10-18): Release v1.3.1 (#64)

* fixing version number

* fix #62 : declare DiscordMessage thread as daemonic to avoid trouble with OctoPrint

* fix #61, #63: update readme for 1.3.1 release
PR #67 (2023-05-31): Release v1.3.1 (#64)
* fixing version number
* fix #62 : declare DiscordMessage thread as daemonic to avoid trouble with OctoPrint
* fix #61, #63: update readme for 1.3.1 release
closed 2022-10-18 by bchanudet