bchanudet/OctoPrint-Octorant #63: OctoRant constantly asks to be updated to 1.3.0.2 due to tag and version mismatch

foosel, 2022-10-18:
As mentioned in #62, setup.py says 1.3.0, the GitHub release is tagged as 1.3.0.2, and that mismatch makes OctoPrint ask to update OctoRant again and again.

You should always make sure that the tag of the latest release on GitHub matches the version stated in setup.py, or this will happen.

bchanudet, 2022-10-18:
Yeah I tagged it 1.3.0.2 to fix #61, as the 1.3.0 tag was already existing, but I guess that didn't do any good.

I will release a proper v1.3.1 this evening, hopefully that'll do it.

foosel, 2022-10-18:
Got the record, you can usually just easily delete a tag and recreate it. Have had to do this a bunch of times over OctoPrint's lifespan while preparing an update.

closed 2022-10-18 by bchanudet