bchanudet/OctoPrint-Octorant #68: Not sendindg messages after reinstall

LukasBlk, 2023-06-02:
Hi, I'm having a problem with Discord notifications. 

I had Octorant installed on my Octoprint RPi server for almost 2 years and it was connected to my Ender 3 Pro, Octorant was sending all the messages to my Discord server. There was no issue at all. After 2 years I bought a new Ender S1 Pro and decided to reinstall Octoprint so that each printer has its own instance of Octoprint. I used this script (https://github.com/paukstelis/octoprint_deploy). 

Anyway, after installing and setting up both printers and installing Octorant on each instance of Octoprint, I'm now experiencing this issue. Neither of the printers is sending messages to my discord server about starting, pausing, or finishing print. Even for the Ender 3 Pro which worked 2 years before, I did not change the firmware, the only new thing is a new version of Octoprint and Octorant.  I receive only Octoprint online message and Printer ready message, after that, no other info about printing, pausing, and finishing prints is not sent to Discord. 

Both of the printers are now showing this message in OctoPrint ("Your printer's firmware supports host action commands, but they are disabled."), but before the installation process, the Ender 3 was totally working. 

I googled several days but not found anyone else having the same issue, Can you please help me? If you need some log files for the printers, I will add them according to your instructions. 

[image]
[image]

[image]

bchanudet, 2023-06-03:
Hi,

Could you maybe provide your server logs? Usually there is at least one Octorant line for each message that should be sent, so we should be able to see what happened.

If you don't want to attach them here for privacy reasons, you can also send them direct to me to dev@benjaminchanudet.com

LukasBlk, 2023-06-03:
Hi,

thank you for your message. Can you help me a little bit about what log you need? I don't have many programming skills. I'm running OctoPrint with Octorant on Raspberry Pi and OctoPi OS.

bchanudet, 2023-06-03:
Don't worry, that doesn't need programing skills. :)

Open OctoPrint settings. Under the "OctoPrint" section, you should find a page for "Logging". In there, you'll see a list of log files. The one simply named "octoprint.log" is the one I'm interested in.

Download the file, then when answering to the thread here, you will be able to drag & drop the file into the message to attach it automatically.

LukasBlk, 2023-06-03:
Thank you for the instructions, here is the LOG file from the instance with Ender S1 Pro connected.

[S1PRO_octoprint.log](https://github.com/bchanudet/OctoPrint-Octorant/files/11643380/S1PRO_octoprint.log)

bchanudet, 2023-06-03:
Many thanks for the logs. 

From what I'm seeing you are affected by a bug that I also encountered myself last week, and that should already be fixed by cd268fe877010bbae77d0903166ebd9d5f951554

I'm hopeful to have a new version to release this weekend or beginning next week. Once done I'll probably message you here to confirm that this fixes your bug. :)

LukasBlk, 2023-06-04:
Thank you for the diagnostic and help :) I will test my setup once you release the new version and give feedback.

bchanudet, 2023-06-05:
I just released the v1.3.2 of the plugin, feel free to test it and report back to me if this fixed your bug. :)

LukasBlk, 2023-06-17:
Hi,

I finally updated your plugin and it seems nothing has changed. I
uninstalled it, restarted the Raspberry, and installed the new plugin
again. Still no info messages about printing, just starting of octoprint
and printer ready :(

[image: image.png]
[image: image.png]

bchanudet, 2023-07-02:
Hi,

Sorry for this. 

Same as the previous time, could you maybe provide some logs that would help me diagnose this case?

Many thanks :)

LukasBlk, 2023-07-03:
Hi,

sure thing, here's the current log file :)

[image: image.png]

bchanudet, 2023-07-03:
Sadly, replies by email don't support attachments, so I see nothing in your last reply.

LukasBlk, 2023-07-05:
Uploading log via GitHub again :)

[image]

[octoprint.log](https://github.com/bchanudet/OctoPrint-Octorant/files/11954787/octoprint.log)

bchanudet, 2023-07-06:
Hello,

@LukasBlk, I don't see in your log any prints running, therefore I can't see what errors OctoRant triggers (but I can confirm it loads correctly during OctoPrint startup. Can you try printing a small object to see if we can detect errors. 

@Wackeljupp, same thing, could you provide your octoprint logs? (best would be those generated during a print)

Also to you both, can you give me your OctoPrint version, and how it's installed? (via OctoPi, OctoPi with the new camera stack, or by manual installation?)

Wackeljupp, 2023-07-15:
Sorry for late reply.
Here is my log from a print.
I can't really tell how i installed it. 
[octoprint (3).log](https://github.com/bchanudet/OctoPrint-Octorant/files/12060407/octoprint.3.log)

LukasBlk, 2023-07-16:
I also apologize for the late reply, here's my log file. I'm running OctoPrint version 1.9.0, Python 3.9.2 and I installed it via Raspberry Pi Imager, I selected the OctoPi installation which includes OctoPrint.
[octoprint.log](https://github.com/bchanudet/OctoPrint-Octorant/files/12063812/octoprint.log)

Wackeljupp, 2023-07-19:
B/c of another problem i reinstalled everything now.
and now it works properly

commit 7f5f245e (2023-08-02): Hopefully fixes #68

- Use a more native type "queue.Queue()"
- Move the start of the thread in the constructor to make sure it's called only once.
commit dd5450b0 (2023-08-08): Hopefully fixes #68

- Use a more native type "queue.Queue()"
- Move the start of the thread in the constructor to make sure it's called only once.
commit 09d072ea (2023-08-18): Release 1.3.3 (#72)

* Hopefully fixes #68

- Use a more native type "queue.Queue()"
- Move the start of the thread in the constructor to make sure it's called only once.

* reduce deprecation notices in logs by better handling legacy settings access

* Update version for release
closed 2023-08-18 by bchanudet