bchanudet/OctoPrint-Octorant #75: Also allow Cura thumbnails

BobSleeuw, 2023-10-21:
Is it possible to also allow .ufp thumbnail files for gcode sliced in Cura? Other plugins like Slicer Thumbnails will show the Cura thumbnails, but that still doesn't make it possible to use them in this plugin.

bchanudet, 2023-10-22:
Hello,

I'm not familiar with Cura's thumbnails. Are they generated aside the gcode or inside like PrusaSlicer? 

Is there a way for you to send me a sample here on this issue? Many thanks :)

BobSleeuw, 2023-10-24:
They are generated in the gcode, as commented lines at the start (I believe with base64 encoding). They are denoted at the start by `; thumbnail begin width height <variable I don't know>` and end at `; thumbnail end`. 

Here is a sample .gcode file of a small cube (I changed the extension to .txt since github doesn't support .gcode)

[5mmCube.txt](https://github.com/bchanudet/OctoPrint-Octorant/files/13111418/5mmCube.txt)

bchanudet, 2023-10-24:
Thanks for the sample gcode!

That seems awfully close to the format I already handle for PrusaSlicer's thumbnails, I'll have a look at what's different between the two and check. :)

BobSleeuw, 2023-10-24:
Awesome, thanks!

closed 2023-11-05 by bchanudet

BobSleeuw, 2023-11-08:
Works like a charm, awesome! Thanks for the quick fix @bchanudet!

commit 616ba696 (2023-11-05): Release 1.3.4 (#76)

* fix threshold that was reaaally low
* fix #75
* Release 1.3.4