bchanudet/OctoPrint-Octorant #78: Webhook Not Functioning/Sending

SalmonSays, 2024-05-03:
Hey, I'm very hopeful for advice regarding Octorant not sending messages. I'm hitting a wall troubleshooting, in that I keep seeing "octoprint.plugins.octorant - INFO - Settings have changed. Send a test message..." in the octoprint logs but not much else: 
[octoprint.txt](https://github.com/bchanudet/OctoPrint-Octorant/files/15205162/octoprint.txt)

I have two identical raspberry pi 3B+'es, one has a webcam connected and works fine (Octopi 0.18, octoprint 1.10.0). The second pi (without a webcam) refuses to send messages. I tried temporarily changing the Stream URL from a long-closed github issue. I've gone so far as to re-image the second pi to the newest Octopi 1.0.0 (octoprint 1.10.0) and even after a factory reset with only the OctoRant plugin freshly installed it still won't send messages to Discord using webhooks. The webhook URL is identical between devices, though I've tried a fresh/new webhook as well to be thorough, and I've manually disabled all optional screen/snapshot uploads from events in OctoRant. I also know both devices have working internet access (I can update plugins, etc).

I have no idea how/where to troubleshoot this next myself, so any input would be greatly appreciated.

ArcAIN6, 2024-05-03:
Same issue here... Wonder what's causing it.

SalmonSays, 2024-05-06:
I bought a cheap Logitech C270 for $15 to mount on my Mini+ (https://www.printables.com/model/125065-better-logitech-c270-prusa-mini-mount) and suddenly OctoRant is working fine. I feel like it shouldn't absolutely require a camera to work properly, but I got tired of fighting with the camera-less setup and figured I'd be installing one eventually anyway. All the webhook variants I set up are functioning exactly as intended on the first try now.

ArcAIN6, 2024-05-07:
I think my problem might be a little different, i have two cameras, the raspberry pi camera module, and a usb camera, both work, and are usable by octoprint, but there's not way to select which one for  octorant, so i'm wondering if it's getting confused *shrugs*

bchanudet, 2024-05-08:
Hello,

Thanks for reporting this issue and including so much details. 

I'll have a look on the code and hopefully will be able to release a fix soon.