bchanudet/OctoPrint-Octorant #79: Octorant will not post any messages if webcam snapshot is enabled with new camera stack

ArcAIN6, 2024-05-09:
I would like to preface this with a bit of information about the setup. 

 - There are 2 cameras attached to the Raspberry Pi 3B+, one is a USB webcam, the other is a Raspberry Pi Camera Model 2 (CSI connections)
 - I also have a plugin called "Multicam" installed (Not sure if this would affect it, but it's the only other camera related plugin i have installed)
It allows me to utilize both cameras in the dashboard (https://github.com/mikedmor/OctoPrint_MultiCam)
 - Before upgrading to octoprint 1.1.0 with the new camera stack. Octorant worked perfectly. 
 - The only changes i've made to the system is upgrading to 1.1.0 with the new camera stack, added the webcam, and added multicam. 

Unfortunately, there are no "ERROR", "WARNING"", or "INFO"'s in octoprint's log files. so trying to determine exactly what has gone wrong where is anyone's guess. 

Directly access the cameras via URL functions, albeit a bit different for each camera. 

Raspi-Cam:
http://[LOCAL-IP]/webcam/?action=stream
http://[LOCAL-IP]/webcam/?action=snapshot

USB-Cam: 
http://[LOCAL-IP]:8082/?action=stream
http://[LOCAL-IP]:8082/?action=snapshot

As you'll notice, the USB Camera doesn't use the /webcam/ subfolder in the URL, that's because it hasn't been configured through HAProxy (Still haven't found a working tutorial on how to do that) 

The cameras both function within octoprint as expected, and all four URL's produce media (1 stream, and 1 snapshot per camera)

Looking at the documentation, as well as the media.py here, it appears there are different methods being used for pulling the camera. 
In media.py, 

```from octoprint.webcams import get_snapshot_webcam```

From https://docs.octoprint.org/en/master/plugins/mixins.html#webcamproviderplugin :

[image]

Maybe there has been a change in the way cameras are handled, for plugins as well, i'm unsure, as i haven't looked to build my own plugin yet XD

bchanudet, 2024-05-09:
Hello,

Many thanks for all the details. At first glance it seems related to the other bug reported in #78.

By default OctoRant uses the camera set up as the snapshot camera in the "Webcam & Timelapse" section of the settings. Could you tell me which camera is selected as the "snapshot webcam", as well as the "fallback webcam"?

ArcAIN6, 2024-05-09:
Tried both USB camera,as well as the pi camera, didn't seem to make a difference, but currently have the USB camera as the default in that setting:
[image]

bchanudet, 2024-05-09:
Okay so first results of investigations point towards Multicam's fault.

In the [WebcamProviderPlugin docs](https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.WebcamProviderPlugin), `take_webcam_snapshot()` is supposed to be a string of the camera name given by the configuration. Whereas [Multicam's implementation](https://github.com/mikedmor/OctoPrint_MultiCam/blob/277501232100dd033020f25e6229e1f250b6cdcf/octoprint_multicam/__init__.py#L159) seems to expect a `Webcam` object, as the first operation is to extract the settings from the `provided_webcam` parameter.

I'll raise an issue on the Multicam repository to see with the plugin author if we can find something that works for both of us. Hopefully this will be resolved soon.

ArcAIN6, 2024-05-09:
cool :) Thanks for investigating <3